nvidia-isaac/cuVSLAM #1: invalid ELF header

Yasintrkmn, 2025-03-26:
When I try to run `python3 examples/kitti/track_kitti.py` on ubuntu, I got following
`ImportError: pycuvslam/cuvslam/x86/cuvslam/pycuvslam.so: invalid ELF header`

jbiswas-nvidia, 2025-03-27:
Hi @Yasintrkmn thanks for trying out PyCuVSLAM.
To help us investigate the issue, can you please tell us what your OS and platform is?

jbiswas-nvidia, 2025-03-27:
Could you please upgrade CUDA to 12.6, and try again? 
CUDA 11.x is not supported - https://github.com/NVlabs/PyCuVSLAM?tab=readme-ov-file#system-requirements

Yasintrkmn, 2025-03-28:
OK. Thanks

closed 2025-03-28 by Yasintrkmn

hchlhwang, 2025-04-02:
I am getting the same issue with the cuda 12.6 when using `pip install -e cuvslam/aarch64.` Could I get any advice on this? The following are my system specs:

Jetson AGX Orin
OS: Ubuntu 22.04 
Platform: aarch64
Python: 3.10.13
Cuda: 12.6
Jetpack: 6.2+b77