nvidia-isaac/cuVSLAM #3: Ubuntu 20.04 support

zhanghua7099, 2025-04-09:
Hi! Thanks for open-sourcing this work.

I'm trying to use this project on **Ubuntu 20.04**, and I would really appreciate any advice or suggestions you might have.

I've tried both **Conda** and **Docker**, but have run into some issues:

* With **Conda**, I encountered the following import error:
    
    ```
    ImportError: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by /home/robot/PyProject/PyCuVSLAM/cuvslam/x86/cuvslam/pycuvslam.so)
    ```
    
* As for **Docker**, it doesn't seem to work properly when using **Intel RealSense** — specifically, the **IMU data is not accessible** from within the container. This seems to be a known issue, as mentioned here: [https://github.com/IntelRealSense/librealsense/issues/13806#issuecomment-2697150049](https://github.com/IntelRealSense/librealsense/issues/13806#issuecomment-2697150049)
    

Do you have any recommendations for how to best set this up on **Ubuntu 20.04**?

Thanks in advance for your help!

hrabeti-nvidia, 2025-04-10:
Hi Huayan,

Thank you for trying PyCuVSLAM.

Q1) We're currently looking to expand our supported platforms but Ubuntu 20.04 might be too old for us to actively support. Can you upgrade your Ubuntu build, upgrade your glibc, or use a docker image?

Q2) Can you try Stereo-only (Not inertial)? cuVSLAM is quite robust even without IMU data.

Q3) Please refer to https://nvlabs.github.io/PyCuVSLAM/api.html#cuvslam.Rig. You should be able to see how we populate this field in the various examples.

zhanghua7099, 2025-04-11:
Hi @hrabeti-nvidia,

Thank you very much for your helpful responses!

Regarding Q1, I was able to resolve the issue by using Docker. I referenced the following scripts:

* [https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common/blob/main/docker/Dockerfile.realsense](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common/blob/main/docker/Dockerfile.realsense)
    
* [https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common/blob/main/docker/scripts/build-librealsense.sh](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common/blob/main/docker/scripts/build-librealsense.sh)
    

I've made some modifications and pushed them to a separate branch here:  
👉 [https://github.com/zhanghua7099/PyCuVSLAM/tree/rs-support](https://github.com/zhanghua7099/PyCuVSLAM/tree/rs-support)

To try it out, simply run:

```
./run_docker.sh
cd examples/realsense
```

Both `python3 run_vio.py` and `python3 run_stereo.py` are working with **realsense d435i** camera. However, a few parameters may need to be fine-tuned for optimal performance. 

If this is of interest, I’d be happy to submit a PR. Please let me know!

closed 2025-04-12 by zhanghua7099

sidtalia, 2025-07-09:
I'm trying to use this setup in ubuntu 20.04 with python 3.8. Is there a "hard dependency" on 3.8 or is it a thing that can be achieved with minor syntax changes?