nvidia-isaac/cuVSLAM #7: Add support for 5090/blackwell cards

pablovela5620, 2025-06-11:
Blackwell cards dont' seem to work with this implementaiton. Is there any plans to add support?

vikuznetsov-nvidia, 2025-06-16:
Hi Pablo,

Thanks for your interest in our work!

Does it produce any output on 5090/blackwell? Does it work for you on older cards?

Also, what does `nvidia-smi` show?

pablovela5620, 2025-06-16:
Right now the error I'm getting is 
```
(PyCuVSLAM) pablo@pablo-dl-server:~/0Dev/repos/PyCuVSLAM$ python examples/euroc/track_euroc.py 
Traceback (most recent call last):
  File "/home/pablo/0Dev/repos/PyCuVSLAM/examples/euroc/track_euroc.py", line 7, in <module>
    import cuvslam as vslam
  File "/home/pablo/0Dev/repos/PyCuVSLAM/.pixi/envs/default/lib/python3.10/site-packages/cuvslam/__init__.py", line 3, in <module>
    from .pycuvslam import *
ImportError: /home/pablo/0Dev/repos/PyCuVSLAM/.pixi/envs/default/lib/python3.10/site-packages/cuvslam/pycuvslam.so: invalid ELF header
```

I haven't had a chance to try on an older card

nvidia-smi
```
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 570.153.02             Driver Version: 570.153.02     CUDA Version: 12.8     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 5090        Off |   00000000:01:00.0  On |                  N/A |
|  0%   50C    P8             23W /  600W |     776MiB /  32607MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
                                                                                         
+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI              PID   Type   Process name                        GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|    0   N/A  N/A            5182      G   /usr/lib/xorg/Xorg                      279MiB |
|    0   N/A  N/A            5986      G   /usr/bin/gnome-shell                     30MiB |
|    0   N/A  N/A            6581      G   ...exec/xdg-desktop-portal-gnome         58MiB |
|    0   N/A  N/A            7765    C+G   /usr/bin/warp-terminal                   62MiB |
|    0   N/A  N/A           10023      G   /opt/google/chrome/chrome                 6MiB |
|    0   N/A  N/A           10074      G   ...ersion=20250613-130015.573000         36MiB |
|    0   N/A  N/A           62057      G   ...ess --variations-seed-version         33MiB |
|    0   N/A  N/A         1210103      G   /usr/share/code/code                     61MiB |
|    0   N/A  N/A         3810763      G   /usr/share/rustdesk/rustdesk             69MiB |
+-----------------------------------------------------------------------------------------+
```

vikuznetsov-nvidia, 2025-06-17:
This looks more like an incorrect binary format. You need to have `git lfs` installed, then re-clone the repo or try `git lfs pull`. Does it help? 
We'll update the readme soon to include troubleshooting.