nvidia-isaac/cuVSLAM #16: What is the problem using 4-stereo on a Jetson AGX Orin?

C-H-Chien, 2025-09-18:
Hi! Thanks for the amazing work! While reading the paper, I noticed that in Table 1, the 4-stereo setup is "NA" when running on the Jetson AGX Orin. The paper however did not mention/explain that. I would like to know why it is NA, since in Appendix A.2, the hardware utilization for both CPU and GPU are less than 40% when using 3 stereo pairs and thus I guess when scaling up to 4 stereo pairs it is still workable.

Another two questions: (i) is it possible to run 4 stereo pairs on Jetson Orin Nano with HD or Full HD images? (ii) I noticed that Appendix A.1.6 describes the performances using four stereo pairs. Is it running on the Desktop or a Jetson AGX Orin?

aefitorov-nvidia, 2025-09-19:
Hi, thanks for your interest in our work and for the detailed questions!  

**1) 4 stereo-camera support on Jetson AGX Orin**  
From the cuVSLAM core library perspective, there is no limitation preventing real-time processing (30 FPS) of 4 stereo pairs on Jetson AGX Orin. This setup is demonstrated in the [Isaac ROS Nova Carter robot example](https://nvidia-isaac-ros.github.io/v/release-3.1/concepts/visual_slam/cuvslam/tutorial_multi_hawk.html) using 4 FHD Hawk stereo cameras. The corresponding launch file can be found [here](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_examples/blob/897a10568adcf63dd39a9a4430981194050944a9/isaac_ros_multicamera_vo/launch/isaac_ros_visual_slam_multihawk_carter.launch.py#L227).  
The same setup was used for the results in Appendix A.1.6, which are intended to be identical in performance on both Jetson AGX Orin and x86 systems.

**2) Running 4 stereo pairs on Jetson Orin Nano (HD/FHD)**  
The main bottleneck in multi-stereo setups is not GPU/CPU load, but reliably delivering *synchronized batches of images* to the cuVSLAM core with low latency and no drops. Even on Orin AGX, this can be challenging without optimizations - hence the use of the [Nitros](https://nvidia-isaac-ros.github.io/concepts/nitros/index.html) image format ([code reference](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_visual_slam/blob/19be8c781a55dee9cfbe9f097adca3986638feb1/isaac_ros_visual_slam/src/visual_slam_node.cpp#L113)) instead of standard ROS 2 images.  

For Jetson Orin Nano, achieving this level of synchronized delivery with 4 HD stereo cameras over USB is unlikely in practice.  

From a practical standpoint, visual tracking quality is a trade-off between resolution and FPS. For cuVSLAM, VGA (640×480) resolution is sufficient for robust tracking, and increasing resolution generally does not yield major improvements in pose estimation accuracy. FPS is more important - especially for avoiding motion blur and significant displacement of features in successive frames - so we recommend 30 FPS minimum, ideally 60 FPS, depending on motion speed.

C-H-Chien, 2025-09-19:
Thank you so much for the detailed response! I really appreciate it.
So I guess the NA in Table 1 for 4-stereo settings on Jetson AGX Orin are simply missing data, not necessarily mean that they are not achievable. I would love to try the code with multiple cameras but not from Intel RealSense or Hawk stereo cameras, with dedicated hardware to handle frame-level synchronization. Thank you again!

closed 2025-09-19 by C-H-Chien